chirpstack/chirpstack-gateway-bridge #221: ConfigMap integration.mqtt.auth.generic.servers value is not used

NunoSav, 2023-01-31:
- [x] The issue is present in the latest release.
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-gateway-bridge) of this repository and believe that this is not a duplicate.

## What happened?
GW Bridge tries to connect to mqtt broker over at `127.0.0.1` instead of whatever value(s) is in the CM

## What did you expect?
For GW Bridge to use value in ConfigMap

## Steps to reproduce this issue

Steps:

1. Deploy GW Bridge into cluster with config map

## Your Environment

| Component           | Version |
| --------------------| ------- |
| Gateway Bridge      |    4.0.5     |
| Chirpstack API      |    4.2.0     |

## Note
I went back to version 4.0.1 and issue is gone.

eelars, 2023-02-11:
Same issue here but i can connect correctly to the MQTT server in 4.0.3, the issue seems to starts with Chirpstack-gateway-bridge 4.0.4 version.

brocaar, 2023-02-14:
This should be fixed in v4.0.5 which was released shortly after the v4.0.4 release.
See: https://www.chirpstack.io/docs/chirpstack-gateway-bridge/changelog.html#v405.

brocaar, 2023-02-14:
I have just re-tried, but I can not reproduce this issue. With v4.0.4 there was indeed an issue, but I believe it was fixed by v4.0.5.

> Deploy GW Bridge into cluster with config map

I'm not really sure what this means.

I tried to reproduce the issue by changing:

```toml
    [integration.mqtt.auth.generic]
    # MQTT server (e.g. scheme://host:port where scheme is tcp, ssl or ws)
    servers=["tcp://localhost:1883"]
```

config section to something different and it correctly picked up the new configuration. As well I tried setting the config using an environment variable, and it also picked up this value correctly:

```
❯ env INTEGRATION__MQTT__AUTH__GENERIC__SERVERS=tcp://foobar:1883 ./build/chirpstack-gateway-bridge
INFO[0000] starting ChirpStack Gateway Bridge            docs="https://www.chirpstack.io/gateway-bridge/" version=4.0.5
INFO[0000] backend/semtechudp: starting gateway udp listener  addr="0.0.0.0:1700"
INFO[0000] metrics: starting prometheus metrics server   bind="127.0.0.1:3030"
ERRO[0000] [client]   dial tcp: lookup foobar on 10.0.0.1:53: no such host  module=mqtt
WARN[0000] [client]   failed to connect to broker, trying next  module=mqtt
ERRO[0000] [client]   Failed to connect to a broker      module=mqtt
ERRO[0000] integration/mqtt: connection error            error="network Error : dial tcp: lookup foobar on 10.0.0.1:53: no such host"
```

closed 2023-02-14 by brocaar

NunoSav, 2023-02-14:
> > Deploy GW Bridge into cluster with config map
> 
> I'm not really sure what this means.

I'm deploying Chirpstack inside a k3s cluster and i inject it's configuration through a ConfigMap. With version 4.0.5 running the content of said ConfigMap is ignored.

brocaar, 2023-02-14:
Please note that k3s/k8s does a lot of abstraction for you including handing configuration through ConfigMaps. To understand your issue better, could you please simplify it to something I or somebody else can reproduce, without setting up a k3s/k8s cluster?

ChirpStack Gateway Bridge has two ways to receive configuration:

* Through a configuration file
* Through environment variables

As I explained in my previous comment, both work for me.

NunoSav, 2023-02-14:
Fair enough, I'll do that when I get some free time and come back to you. Cheers