chirpstack/chirpstack-gateway-bridge #230: Add bcning field in router_config response

soorajm13, 2023-05-24:
- [ ] I have searched the [issues](https://github.com/brocaar/chirpstack-gateway-bridge) of this repository and believe that this is not a duplicate.

## Summary

The bcning is currently not sent in the router_config response.

## What is the use-case?

Class B will require enabling bcning support. Tested with the currently released Chirpstack V4 docker, couldn't enable class B beacons and test it.

Example gateway configuration: Chirpstack V4 docker + LoRa basicstation 2.0.6 + sx1302 concentrator

## Implementation description

Add bcning field to the LNS response. Update documentation and add news steps for enabling class B beacon support.   

## Can you implement this by yourself and make a pull request? No

kairoswater-jason, 2024-08-29:
I (as an end user) just reached the same conclusion for my own application
While I suspect I could hack my own instance of the bridge to send hard-coded beacon information. It would be better to get the beacon configuration information from chirpstack rather than hardcode it.

kairoswater-jason, 2024-08-29:
For my reference, any anyone elses

source in basicstation that parses that beaconing field: (which is part of router_config)
https://github.com/lorabasics/basicstation/blob/ba4f85d80a438a5c2b659e568cd2d0f0de08e5a7/src/s2e.c#L1100

example of a configuration used to test basicstation:
https://github.com/lorabasics/basicstation/blob/ba4f85d80a438a5c2b659e568cd2d0f0de08e5a7/regr-tests/test3d-bcns/test.py#L115

```
conf['bcning'] = {
 'DR': 8,
 'layout': [5,11,23],
 'freqs': [923300000 + chx * 600000 for chx in range(8)]
}
```