chirpstack/chirpstack-gateway-bridge #233: Chirpstack v4.6.0 missing uplink info latest gw bridge tag

ccall48, 2024-01-24:
- [x] The issue is present in the latest release.
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-gateway-bridge) of this repository and believe that this is not a duplicate.

## What happened?
I run chirpstack using chirpstack-docker containers and repo. On trying to run latest tag on chirpstack v4.6.0 I get the following errors on the gateway bridge with no data being transmitted.

## What did you expect?
Latest to be compatible with v4.6.0 / as this is whats pulled from docker with container tag 4

## Steps to reproduce this issue
Steps:

1. replace gateway bridge from 4 to latest in docker compose

## Could you share your log output?

This is a snippet of one error, as the logs are full of these for every uplink from all gateways on every region.
```shell
chirpstack-docker-chirpstack-1        | 2024-01-12T08:28:56.977789Z  INFO chirpstack::gateway::backend::mqtt: Message received from gateway region_id="au915_1" topic="au915_1/gateway/2864afc74baa57f1/event/up" qos=0 json=false
chirpstack-docker-chirpstack-1        | 2024-01-12T08:28:57.180429Z  WARN chirpstack::uplink: tx_info of uplink event is empty, skipping
chirpstack-docker-chirpstack-1        | 2024-01-12T08:28:57.180469Z  WARN chirpstack::uplink: tx_info of uplink event is empty, skipping
chirpstack-docker-chirpstack-1        | 2024-01-12T08:28:57.180702Z ERROR chirpstack::uplink: Deduplication error error=Unable to get first item from rx_info
```

## Your Environment
This is docker-chirpstack running on ubuntu cloud server, built new as version 4 - so not an update from v3.

Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.6 LTS
Release:	20.04
Codename:	focal

brocaar, 2024-01-25:
I tried to reproduce:

* Cloned latest `master` version
* Did a docker compose pull
* docker compose up
* Connected (eu868) gateway
* Tested an OTAA
* Device activated fine

```
chirpstack-gateway-bridge-1               | time="2024-01-25T13:10:59.016411521Z" level=info msg="integration/mqtt: downlink frame received" downlink_id=4138928630 gateway_id=0016c001ff1a0a5f
chirpstack-gateway-bridge-1               | time="2024-01-25T13:10:59.023083574Z" level=info msg="integration/mqtt: publishing event" event=ack qos=0 topic=eu868/gateway/0016c001ff1a0a5f/event/ack
chirpstack-1                              | 2024-01-25T13:10:59.023485Z  INFO chirpstack::gateway::backend::mqtt: Message received from gateway region_id="eu868" topic="eu868/gateway/0016c001ff1a0a5f/event/ack" qos=0 json=false
chirpstack-1                              | 2024-01-25T13:10:59.023742Z  INFO tx_ack{downlink_id=4138928630}: chirpstack::downlink::tx_ack: Log downlink-frame for gateway gateway_id=0016c001ff1a0a5f
chirpstack-1                              | 2024-01-25T13:10:59.024049Z  INFO tx_ack{downlink_id=4138928630}: chirpstack::downlink::tx_ack: Log downlink-frame for device device_eui=0101010101010101
```

ccall48, 2024-01-26:
I wonder if the issue is present only if the original build is done using gwb container tagged 4.
`docker compose down --remove-orphans` 
swap the gwb with the latest tag and brought back up
`docker compose up`
all the existing registered devices coming from gwmp show the gw tx/rx errors which is about 4 regions atm.

I will see if i can reproduce on my homelab this week as the issue is occurring on my production server.

ccall48, 2024-01-26:
Seems to reproduce straight away on my testlab with the above, I even created a new us915_1 bridge then added 2x new virtual devices and tried to join them.

Here is a pastebin to the logfile.
https://pastebin.com/hRsRPj2f

PhilDay-CT, 2024-02-24:
I'm seeing the same thing on my setup

Chipstack server is v4.6.0
Gateway Bridge (running in a Lorix 1)  is v3.14.8

GW bridge logs shows:
```
udp-packet-forwarder: JSON up: {"rxpk":[{"tmst":2207589700,"chan":2,"rfch":1,"freq":868.500000,"stat":1,"modu":"LORA","datr":"SF12BW125","codr":"4/5","lsnr":6.5,"rssi":-61,"size":34,"data":"QEL3WwGAFhQF5AQx8yq4e9njuKLeMAnwlynxlAlE562l1w=="}]}
```

Chirpstack Log shows:
```
2024-02-24T09:22:44.595765Z TRACE chirpstack::uplink: Adding uplink event to deduplication set key="up:collect::40d72ac80080af18052f89c719d5d4a3844c5b2748997854bec6"
2024-02-24T09:22:44.596493Z TRACE chirpstack::uplink: Requesting deduplication lock lock_key="up:collect::40d72ac80080af18052f89c719d5d4a3844c5b2748997854bec6:lock"
2024-02-24T09:22:44.597242Z TRACE chirpstack::uplink: Waiting for more uplink events to receive key="up:collect::40d72ac80080af18052f89c719d5d4a3844c5b2748997854bec6"
2024-02-24T09:22:44.798930Z TRACE chirpstack::uplink: Collecting received uplink events key="up:collect::40d72ac80080af18052f89c719d5d4a3844c5b2748997854bec6"
2024-02-24T09:22:44.799417Z  WARN chirpstack::uplink: tx_info of uplink event is empty, skipping
2024-02-24T09:22:44.799564Z ERROR chirpstack::uplink: Deduplication error error=Unable to get first item from rx_info

```

brocaar, 2024-02-26:
Have you tried with the `:4` tag? I think that is a lot safer to use.

brocaar, 2024-03-28:
@ckvi the title of this issue is the problem with the `:latest` tag. Please note that this is something different than the `:4` tag.

ckvi, 2024-03-29:
I see. I tried pulling the v4 Branch and with that it worked immediately. Thanks for your reply.

closed 2024-04-16 by brocaar