chirpstack/chirpstack-gateway-bridge #237: Chirpstack-gateway-bridge, gateway 16 channels (basic station, SX1302, SX1303)

BlokhinVega, 2024-05-25:
Hi all!

Does chirpstack-gateway-bridge support the ability to connect Gateways built on the reference SX1302, SX1303?

The example configuration file itself says that it is intended for the SX1301.

Currently testing a Gateway prototype with two SPI LoRa modules: 8 + 8 channels

I want to connect the gateway to the server as one 16-channel device, and not as two 8-channel devices..

Has anyone experienced this possibility?

BlokhinVega, 2024-05-27:
It’s probably worth clarifying that I use the LoRa Basic Station backend on the gateway :)

sp193, 2025-01-04:
From the code in https://github.com/chirpstack/chirpstack-gateway-bridge/blob/master/internal/config/config.go#L43, you should be able to describe multiple concentrators.

Could it work if you specified all the backend.basic_station.concentrators.* sections multiple times? Since the 16-channel gateways have 2x concentrators, you should have 2 sets.