chirpstack/chirpstack-gateway-bridge #240: TLS Support by Reverse-Proxy for Basicstation

bastienvty, 2024-10-03:
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-gateway-bridge) of this repository and believe that this is not a duplicate.

## Summary

This feature allows the basicstation to be aware that TLS support is managed by a reverse-proxy with TLS termination. This means it doesn't have to manage certificates and only starts a basic http server (`ws://`).

## What is the use-case?

I use the reverse-proxy traefik and noticed that the gateways always ask for `router-info` of the basicstation. It then returns an URI with the scheme configured (`ws` or `wss`). However, chirpstack-gateway-bridge is configured to return `wss` only if TLS certificates have been referenced in the configuration file. 
With a reverse-proxy, TLS is terminated at the network edge. The gateway-bridge therefore works with the `ws` scheme inside but is only accessible through `wss` from Internet. So, it would have to return `wss` even if there are no TLS certificates referenced.

## Implementation description

Simply add a configuration variable to notify the gateway-bridge that TLS is managed by a reverse-proxy. It then modifies the scheme accordingly, even though it only starts a simple http server.

## Can you implement this by yourself and make a pull request?
I already implemented it and will create a pull-request in the hope it gets accepted.

JustDr00py, 2025-03-15:
Im having this same issue how did you fix it? does this fix apply to the docker installation as well?

bastienvty, 2025-03-18:
I resolved it with the PR #241. You can use it to see if your problem persists.