chirpstack/chirpstack-gateway-bridge #242: FSK frequency, data rate not configured to basicstation

ywa037, 2024-10-28:
- [ ] The issue is present in the latest release.
- [ ] I have searched the [issues](https://github.com/brocaar/chirpstack-gateway-bridge) of this repository and believe that this is not a duplicate.

## What happened?
{\"msgtype\":\"router_config\",\"NetID\":null,\"JoinEui\":null,\"region\":\"AS923\",\"hwspec\":\"sx1301/1\",\"freq_range\":[915000000,928000000],\"DRs\":[[12,125,0],[11,125,0],[10,125,0],[9,125,0],[8,125,0],[7,125,0],[7,250,0],[0,0,0],[-1,0,0],[-1,0,0],[-1,0,0],[-1,0,0],[-1,0,0],[-1,0,0],[-1,0,0],[-1,0,0]],\"sx1301_conf\":[{\"radio_0\":{\"enable\":true,\"freq\":917700000},\"radio_1\":{\"enable\":true,\"freq\":918700000},\"chan_FSK\":{\"enable\":true},\"chan_Lora_std\":{\"enable\":true,\"radio\":1,\"if\":-100000,\"bandwidth\":250000,\"spread_factor\":7},\"chan_multiSF_0\":{\"enable\":true,\"radio\":0,\"if\":-400000},\"chan_multiSF_1\":{\"enable\":true,\"radio\":0,\"if\":-200000},\"chan_multiSF_2\":{\"enable\":true,\"radio\":0,\"if\":0},\"chan_multiSF_3\":{\"enable\":true,\"radio\":0,\"if\":200000},\"chan_multiSF_4\":{\"enable\":true,\"radio\":0,\"if\":400000},\"chan_multiSF_5\":{\"enable\":true,\"radio\":1,\"if\":-400000},\"chan_multiSF_6\":{\"enable\":true,\"radio\":1,\"if\":-200000},\"chan_multiSF_7\":{\"enable\":true,\"radio\":1,\"if\":0}}],\"MuxTime\":1730103464.535724}"
## What did you expect?
FSK frequency, data rate not configured to basicstation
## Steps to reproduce this issue

Steps:

1.
Modify chipstack gateway-bridge.toml and add FSK configuration，as:
[backend.basic_station.concentrators.fsk]
frequency=919000000
2.
3.
4.

## Could you share your log output?

```shell

```

## Your Environment

| Component           | Version |
| --------------------| ------- |
| Application Server  | v?.?.?  |
| Network Server      |         |
| Gateway Bridge      |         |
| Chirpstack API      |         |
| Geolocation         |         |
| Concentratord       |         |

ywa037, 2024-10-29:
chirpstack-gateway-bridge version 4.0.10

sp193, 2025-01-04:
The logic for setting the centre frequency is missing for FSK mode: https://github.com/chirpstack/chirpstack-gateway-bridge/blob/master/internal/backend/basicstation/structs/router_config.go#L244
```
				c.SX1301Conf[concentratorNum].ChanFSK = SX1301ConfChanFSK{
					Enable: true,
				}
```