chirpstack/chirpstack-gateway-bridge #243: Issue with sending Device Class C Downlinks (Packet Forwarding to Gateway with ChirpStack MQTT Bridge v4)

cbusse1, 2024-11-10:
I have encountered an issue with downlink packet forwarding between a ChirpStack network server and a RAK gateway using the ChirpStack MQTT bridge. Below is a brief and detailed description of the problem, as well as the configuration used during testing.

## **Brief Description:**

I am experiencing an issue where scheduling a downlink message with an **application payload size greater than 177 bytes results in an invalid message** being sent by the gateway. Specifically, the gateway transmits a null-type message with all fields set to null. This issue occurs when the gateway is operating in packet forwarder mode with WisGate OS 2.2.2 and ChirpStack MQTT Bridge v4 (Protobuf).

## **Detailed Description:**

### **Hardware and Configuration Overview:**

- Affected LoRaWAN Gateway Products:
  - RAK Edge Lite (RAK7268CV2)
  - RAK Edge Pro (RAK7289V2)
- **Firmware:** WisGateOS_2.2.2_RAK
- LoRa Configuration:
  - **Packet Forwarder Mode**
  - **Frequency Plan:** Germany, EU868, Conform To LoRaWAN
- Protocol: 
  - LoRa Gateway MQTT Bridge
  - **MQTT for ChirpStack 4.x (Protobuf)**
  - Configuration mostly using default settings

### **ChirpStack Server Configuration:**

- **ChirpStack v4 Network Server** running on a Raspberry Pi 4
- Setup using the ChirpStack Docker Compose (latest commit [[50159c0](https://github.com/chirpstack/chirpstack-docker/commit/50159c0e19f31a5ae692bd42cfe2ff7dce516194)](https://github.com/chirpstack/chirpstack-docker/commit/50159c0e19f31a5ae692bd42cfe2ff7dce516194))

### **Test Scenario:**

- Sending downlink messages to a Class C device with a **payload size larger than 177 bytes.**

**Observations:**

- The issue is consistently reproducible with any application payload size greater than 177 bytes.

- The problem appears independent of the `rx2dr` setting (tested with DR6 and DR4).

- The issue was observed on both RAK Edge Lite and RAK Edge Pro gateways configured in packet forwarder mode using the ChirpStack v4 MQTT bridge.

- Note that when using a RAK WisGate Developer Gateway (RAK7271), I was able to successfully send downlink packets with payload sizes larger than 177 bytes up to the maximum payload size according to LoRaWAN specifications (242 bytes). This also suggests that the issue lies within the ChirpStack MQTT Bridge, because if I understood correctly the Bridge is not used with the Developer Gateway (since it is directly conncted via USB).

  - In brief, for the setup of the developer gateway, I simply run on the raspberry:

  - ~~~
    sudo ./install.sh --chirpstack=not_install
    ~~~

  - and then adjust the ip in `gateway_conf` to localhost so that I could use my Chirpstack docker instance as usual.

**Diagnostics:**

The WisGate Diagnostics tab also suggests an error during parsing of the downlink message from ChirpStack:

~~~
Tue Aug 27 11:13:27 2024 user.warn lora_pkt_fwd[6775]: WARNING: [down] mismatch between .size and .data size once converted to binary
~~~

**Steps to Reproduce:**

1. Set up the ChirpStack server using Docker on a Raspberry Pi according to the official setup instructions.
2. Register an end-device and configure it as Class C for testing (LoRaWAN Spec. 1.0.3A).
3. Schedule a downlink with an application payload size greater than 177 bytes (e.g., 180 or 200 bytes).

**Appendix:** I have attached screenshots and packet captures comparing downlink transmissions with payload sizes of 177 and 178 bytes, both on the ChirpStack web interface and the WisGate gateway.

--------------------------

I reported this issue to [[support@rakwireless.com](mailto:support@rakwireless.com)](mailto:support@rakwireless.com) and posted it on the community forum too ([[Link](https://forum.rakwireless.com/t/issue-with-downlink-packet-forwarding-on-wisgate-os-2-2-2-with-chirpstack-mqtt-bridge-v4/13368)](https://forum.rakwireless.com/t/issue-with-downlink-packet-forwarding-on-wisgate-os-2-2-2-with-chirpstack-mqtt-bridge-v4/13368)) but have not yet received any response. I would appreciate any advice or tips you may have. At least I want to understand whether the problem is within RAK's implementation of the Chirpstack-MQTT-Bridge, the bridge implementation in general, or, the problem may be caused on my side due to some misconfiguration.

Best regards,

Christian

## Appendix

[image]

[image]

[image]

[image]

[image]

[image]

[image]

brocaar, 2024-11-11:
I don't think the WisGateOS firmware has the ChirpStack MQTT Forwarder embedded. They have implemented a forwarder which is compatible with the ChirpStack v4 msg. format. It might be better to report this to RAK.

cbusse1, 2024-11-12:
Thanks for your quick response. I finally got confirmation from RAK that this is a bug in WisGate OS2 which will be fixed in an upcoming release.

closed 2024-11-12 by cbusse1